adafruit/Adafruit-MCP23017-Arduino-Library #10: Allow for disabling of Interrupts

konfu, 2016-01-31:
There is only a function to enable interrupts per pin but none to disable interrupts.

Please offer functions to disable interrupt of a single pin or all pins altogether.

PsuFan, 2021-02-08:
Yes, I agree. Ran into this too... Lost hours because a) MCP wasn't being reset when I was pushing esp8266 reset, b) I wasn't getting the correct interrupt pin and c) this function doesn't exist!

`void Adafruit_MCP23017::disableInterruptPin(uint8_t pin) {
  // disable the pin for interrupt
  updateRegisterBit(pin, LOW, MCP23017_GPINTENA, MCP23017_GPINTENB);
}
`

ladyada, 2021-02-08:
hi please submit a PR with the fix!

PsuFan, 2021-02-08:
Okay, I never submitted a pull before. Hope I did it right, #54

PsuFan, 2021-02-08:
Sort of related: How do we feel about disabling all interrupts on begin()?

ladyada, 2021-02-08:
thats fine

caternuson, 2021-05-27:
Closing. Fixed with #54.