adafruit/Adafruit-MCP23017-Arduino-Library #12: MCP23018 compatibility

intedinmamma, 2016-07-20:
I've tried out the library with the MCP23018 , and it (not surprisingly) seems to work just fine. I don't know if it's necessary to include in the readme, but now it at least is written somewhere. :-)

johntalton, 2018-05-27:
the *18 device has slightly different iocon registers and will not behave as expected (specifically when using interrupts)

caternuson, 2021-05-27:
Yep. So it probably *mostly* works. But could run into issues when hitting those differences.

[image]

[image]

closed 2021-05-27 by caternuson