adafruit/Adafruit-MCP23017-Arduino-Library #15: How to use different device address for MCP23017 so I can use multiple chips?

dpiralis, 2016-11-09:
I want to use more than 3 MCP23017-E/SP DIP28 chips so I can use many digital pins. I tried [this solution ](http://stackoverflow.com/a/36955623/3576279)but only device address with 0x00 is working. The question is what device address should I use for additional chip? 

```
#include "Adafruit_MCP23017.h" 
Adafruit_MCP23017 mcp1; 
#define addr1 0 // tested 1,2,3,4,5, 0x01, 0x02, 0x20, 0x21 but only 0 and 0x00 is working.
void setup() 
  {
    mcp1.begin(addr1); 
    mcp1.pinMode(0, INPUT); //pin 21 on chip
    Serial.begin(9600); 
  } 

  void loop()
  { 
      if(mcp1.digitalRead(0)== HIGH )
      Serial.println("HIGH"); 
  }
```

dpiralis, 2016-11-10:
This works!
```
#include <Adafruit_MCP23017.h>

Adafruit_MCP23017 mcp1; 
Adafruit_MCP23017 mcp2;
/* 

addr 0 = A2 low , A1 low , A0 low  000
addr 1 = A2 low , A1 low , A0 high 001
addr 2 = A2 low , A1 high , A0 low  010
addr 3 = A2 low , A1 high , A0 high  011
addr 4 = A2 high , A1 low , A0 low  100
addr 5 = A2 high , A1 low , A0 high  101
addr 6 = A2 high , A1 high , A0 low  110
addr 7 = A2 high, A1 high, A0 high 111

Connect pin #12 of the expander to Analog 5 (i2c clock)
Connect pin #13 of the expander to Analog 4 (i2c data)
Connect pins #15, 16 and 17 of the expander to ground (address selection)
Connect pin #9 of the expander to 5V (power)
Connect pin #10 of the expander to ground (common ground)
Connect pin #18 through a ~10kohm resistor to 5V (reset pin, active low)

// Input #0 is on pin 21 so connect a button or switch from there to ground
*/
#define addr1 7 // 7 = A1 high, A2 high, A3 high
#define addr2 0 
void setup() 
  {
    Serial.begin(9600); 
    mcp1.begin(addr1);
    mcp2.begin(addr2);
     
    mcp1.pinMode(0, INPUT); //pin 21 on chip
    mcp2.pinMode(0, INPUT); //pin 21 on chip
    
  } 

  void loop()
  { 
      if(mcp1.digitalRead(0)== HIGH )
      Serial.println("HIGH"); 
      delay(1000);
      
      if(mcp2.digitalRead(0)== HIGH )
      Serial.println("HIGH 2"); 
      delay(1000);
  }
```

MarkFromSales, 2019-01-12:
Very helpful, thank you for sharing!

bigpratama123, 2021-01-25:
thanks you for sharing!! 👍

closed 2021-01-25 by ladyada