adafruit/Adafruit-MCP23017-Arduino-Library #19: Expander stops working after a while

sstaub, 2017-12-20:
- Arduino board:  UNO / Nano / Teensy 3.5
- Arduino IDE version: 1.8.5

I'm using following sketch with 2 LEDs on 6 and 7:

`
#include <Wire.h>
#include "Adafruit_MCP23017.h"

Adafruit_MCP23017 mcp;
  
void setup() {  
  mcp.begin();      // use default address 0
  mcp.pinMode(6, OUTPUT);
  mcp.pinMode(7, OUTPUT);
}

void loop() {
  delay(500);
  mcp.digitalWrite(6, HIGH);
  mcp.digitalWrite(7, LOW);
  delay(500);
  mcp.digitalWrite(6, LOW);
  mcp.digitalWrite(7, HIGH); 
}
`
After a short while (2s up to 2min) the LEDs stops blinking. I tried it on different boards, also Teensy 3.5 but can't figure out what the problem is.

ladyada, 2017-12-20:
hiya its likely a wiring issue. please post up in the support forums of wherever you bought your hardware for assistance :)

closed 2017-12-20 by ladyada

sstaub, 2017-12-20:
Thank you, I forgot the reset pin :)