adafruit/Adafruit-MCP23017-Arduino-Library #22: Compatibility with Adafruit Feather M0 (and other SAMD21- boards I guess)

Teletobi91, 2019-01-04:
I cannot get the MCP23017 to work on my Feather board. 

Any ideas, besides using external pullups (already installed)?

I2C scanner doesn't find the chip, Logic Analyzer shows that it doesn't get stuck...

closed 2019-01-04 by Teletobi91

MattGarfield, 2020-03-20:
@Teletobi91 I'm having a similar issue - were you able to resolve this?

Teletobi91, 2020-03-21:
Yes, it had nothing to do with the hardware. 
I found a fault in my PCB. I did't have a defined signal on one of the ENABLE-pins.

MattGarfield, 2020-03-23:
Great, thanks for the update! 

On my end, I found that I needed to put 4.7 kΩ pull-up resistors on the SDA and SCL pins of the MCP23017. The thing that confused me was that everything worked fine _without_ the pull-ups when I used an Adafruit Feather 32u4 with the MCP23017, but it stopped working when I swapped in the Feather M0. I guess the slower clock speed of the 32u4 allowed the MCP23017 to just barely operate without the pull-ups, but the faster clock speed of the M0 made the pull-ups **absolutely necessary**.