adafruit/Adafruit-MCP23017-Arduino-Library #23: Please recognize INPUT_PULLUP for pinmode

euphi, 2019-01-19:
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/blob/1f545fe466a0461c467560820f10fbdb7abe759e/Adafruit_MCP23017.cpp#L130 -->

Change line to 

```C++
	updateRegisterBit(p,(d==INPUT || d==INPUT_PULLUP),MCP23017_IODIRA,MCP23017_IODIRB);

```

INPUT_PULLUP is valid mode for the standard arduino GPIOs, see https://www.arduino.cc/reference/en/language/functions/digital-io/pinmode/ .

The MCP23017 automatically activates its internal pull-UP, so nothing more has to be done.

I prefer to use INPUT_PULLUP as pinmode, because it correspond to the actual behaviour, so it is good practice to document this in the code.

euphi, 2019-01-20:
Done.

After reading the data sheet I'm still unsure if the internal pullups are active by default. So I also added a line to activate them, if the mode is INPUT_PULLUP.

If the mode is not INPUT_PULLUP, the state of pullups is unchanged, so this is no change in behaviour to existing code that uses "INPUT".

The PR also includes debug code to see the "endTransmission" return code. This code is deactivate by default, define `MCP23017_Serial_Debug` to use it.

caternuson, 2021-06-25:
This should be taken care of with the 2.0.0 release:
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/releases/tag/2.0.0
Please update to that and try it out.

Closing for now, but can reopen if still an issue.

closed 2021-06-25 by caternuson