adafruit/Adafruit-MCP23017-Arduino-Library #28: MCP23017: please provide interrupt example also for Adafruit M0, M4, and Arduino Due (M3)!

dsyleixa, 2019-04-15:
please provide interrupt example also for Adafruit M0, M4, and Arduino Due (M3)!
I tried with my dafruit Feather, but it does not compile for that board!

```
exit status 1
'EIFR' was not declared in this scope
```

IoTPanic, 2019-04-25:
I have this library currently working on ItsyBitsy boards (Both M0 and M4 variants) and simply removing this line will cause the example to function just fine. However, while I am clearing the interrupt I use the function  noInterrupts(); and interrupts(); which are built into the Arduino framework that adds some stability to the int clearing.

dsyleixa, 2019-04-25:
but how will MCP23017 and Arduino pinchange interrupts work then for ARM core Arduino boards if the line with 'EIFR' has been removed?
As stated, ported example code versions are needed which provide al  the features not only for AVR but also for ARM boards.

IoTPanic, 2019-04-25:
I don't understand what you're stating, the EIFR register is nonexistant in ARM micro controllers, thats why you get the error, and I have built a full working camera stations with PIR, button, and chip based interrupts  using both PORT interrupts as well as MCP23017 interrupts going into PORTs and everything works great. My understanding is in the more recent arduino frameworks the interrupt flag is cleared automatically during ISR call but this is based on nothing but using the SAMD21 and 51 chips with Arduino past couple months.

dsyleixa, 2019-04-25:
I know, EIFR is a AVR register, used for interrupts, so the code works just for AVRs.
Now a code is needed for ARMs too, providing the same functionality, also to ARM interrupts.
Is that now better understandable?

IoTPanic, 2019-04-25:
_Note, I am not trying to argue, I am giving you info about my finding while using this IC out of the goodness of my damn heart, if you want to disagree that's fine, don't do what I'm explaining._

Okay, and I said, the examples do not need additional code to function on your target, to make it so the examples work in a basic sense, simply remove the line concerning EIFR, and the example will function just like it would on a AVR with this line. Extra stability can be added by turning off all ints while handling then re-enabling them with noInterrupts() and interrupts() respectfully when you're done to prevent additional calls while you're handling it, which is one of the things this line achieves.

As I have explained in a previous post, for ARM micro-controllers in the Arduino framework I believe they clear the EIC during the ISR call for you, since I have been able to use the library for months just how I'm describing it.  These examples were made years ago, when ARM micro-controllers were not popular and the PIC and ATMEGA series AVRs dominated the hobbyist market.

I can make a new pull request with my code to update the examples if you wish that will work out of the box with Arduino, I use the platforms you are describing, and the code works really well without clearing a interrupt bit. Even with other peripherals.

dsyleixa, 2019-04-25:
> I can make a new pull request with my code to update the examples if you wish that will work out of the box with Arduino, I use the platforms you are describing, and the code works really well without clearing a interrupt bit. Even with other peripherals.

yes, thanks, that would be highly appreciated!

IoTPanic, 2019-04-27:
I'm not gonna pull request it since I'm waiting on Adafruit to approve a different one first, but here is a link to my branch https://github.com/IoTPanic/Adafruit-MCP23017-Arduino-Library/blob/ARMified/examples/interrupts_ARM.ino I know the example is in the wrong directory, this is just for you

closed 2021-05-27 by caternuson