adafruit/Adafruit-MCP23017-Arduino-Library #41: get error status

ramarro123, 2020-03-02:
Hello,

it's possible to retrive the current status of OUTPUT pins?
ie: if i do writeGPIOAB(15) can i read pin n*0 and get his value? (or even read back all the status of AB)

if i do readGPIOAB, that's read the actual value on the pin, as INPUT correct? not the prev value set with writeGPIOAB?

caternuson, 2020-03-02:
Should probably work. The `readGPIOAB` function just returns the current values on the ports. It doesn't matter if those values are in or out. Try a simple test to verify - set all to output and low, call read and see what you get, set one pin high, call read and see what you get.

ramarro123, 2020-03-06:
unfortunately reading with readGPIOAB doens't return what's written with writeGPIOAB, not sure why :) but wirth WRITEGPIOAB little leds get on off :D so it work hehe

caternuson, 2020-03-06:
Not sure why either. Just tested and it seems to work.

```cpp
#include "Adafruit_MCP23017.h"

Adafruit_MCP23017 mcp;

void setup() {
  Serial.begin(9600);
  
  mcp.begin();
  for (int p=0; p<16; p++) mcp.pinMode(p, OUTPUT);
}

void loop() {
  Serial.println("----------------------");
  Serial.println("Writing 0b1111111100000000");
  mcp.writeGPIOAB(0b1111111100000000);
  Serial.print("Reading 0b"); Serial.println(mcp.readGPIOAB(), BIN);
  delay(500);
  Serial.println("----------------------");
  Serial.println("Writing 0b1000000011111111");
  mcp.writeGPIOAB(0b1000000011111111);
  Serial.print("Reading 0b"); Serial.println(mcp.readGPIOAB(), BIN);
  delay(500);
}
```

[image]

ramarro123, 2020-03-07:
this code for me
```
	DBG.log("writing %d", bitField);
	expander.writeGPIOAB(bitField);

	bitField = expander.readGPIOAB();
	DBG.log("readed %d", bitField);
```
print

> [0] 07:17:15.427 writing 65407
> [0] 07:17:15.428 readed 65535
>

caternuson, 2020-03-07:
What if you run the example sketch I posted?

caternuson, 2020-03-09:
Closing. Doesn't seem to be a library issue.

closed 2020-03-09 by caternuson