adafruit/Adafruit-MCP23017-Arduino-Library #42: begin as input instead of output

ramarro123, 2020-03-15:


Steinies, 2020-04-15:
Maybe useful. But the user could just enable the pins he need for input. With my changes in #40 you can use multiple devices without losing the settings.

roncoa, 2020-09-05:
In the Adafruit_MCP23017.cpp file, it would be sufficient to replace
	writeRegister(MCP23017_IODIRA,0xff);
	writeRegister(MCP23017_IODIRB,0xff);
with
	// writeRegister(MCP23017_IODIRA,0xff);
	// writeRegister(MCP23017_IODIRB,0xff);
?

Steinies, 2020-09-07:
That would probably work. But that would also change the current default API behavior which could break others code.

rochdi80tn, 2021-01-26:
@ramarro123,
As per my experience,  it is recommended to initialize your MCP as inputs with a predifned values then change them to outputs. This will ensure your desired value is properly set on the output when your MCP starts.

```
mcp.begin(0); // will initialize to inputs
mcp.writeGPIOAB(0xFFFF); // set GPIOAB to desired values here all HIGH

mcp.setPinMode(5, OUTPUT); // --> pin 5 will be for sure at HIGH level
```
Actually I got an issue while linking an MCP with a 8 relay module where I need to driver all outputs high initially to ensure my relays are OFF.
And without my above recommendation, the relay is switched to ON and them to OFF quickly on the startup.. which is not good if you're putting some sensitive device on that relay...

closed 2021-05-27 by caternuson