adafruit/Adafruit-MCP23017-Arduino-Library #43: Can this lib be used  on ATTiny85  ?

sebcaps, 2020-03-22:
I'd like to use this lib on ATtiny85, but while compiling I've got many errors which seems related to Wire.h lib (see below and [here]( https://github.com/nickgammon/I2C_Anything/issues/1
)
`Arduino: 1.8.10 (Windows 10), Board: "ATtiny25/45/85, ATtiny45, Internal 1 MHz" C:\Users\seb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\Wire\src\utility\twi.c: In function 'twi_init': C:\Users\seb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\Wire\src\utility\twi.c:76:16: error: 'SDA' undeclared (first use in this function) digitalWrite(SDA, 1);^~~C:\Users\seb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\Wire\src\utility\twi.c:76:16: note: each undeclared identifier is reported only once for each function it appears in C:\Users\seb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\Wire\src\utility\twi.c:77:16: error: 'SCL' undeclared (first use in this function); did you mean 'SPL'?
digitalWrite(SCL, 1);^~~SPL In file included from c:\users\seb\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\io.h:99:0, from C:\Users\seb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\Wire\src\utility\twi.c:25:C:\Users\seb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\Wire\src\utility\twi.c:80:7: error: 'TWSR' undeclared (first use in this function); did you mean 'ACSR'?
cbi(TWSR, TWPS0);`

 I tried to replace Wire lib with TinyWire with no more success.

Any help appreciated !
Thx

pwm84, 2020-04-23:
In Adafruit MCP23017.h, change lines 17-22 as follows:

// Don't forget the Wire library
//#ifdef __AVR_ATtiny85__           //TinyWireM is now part of
#include <TinyWireM.h>            //   Adafruit version of Wire Library
//#else
//#include <Wire.h>
//#endif

dertobias, 2021-03-29:
also you have to add those lines
after including TinyWireM.h
`#define Wire TinyWireM
#define TwoWire USI_TWI  `

[https://forum.arduino.cc/index.php?topic=656742.0](url)

caternuson, 2021-05-27:
Yep, it should work as long as you can get an I2C library to build for the tiny. Fixing #63 should help with this also, since it will allow whatever I2C bus to be passed in, either `Wire` or `TinyWire` based.

closed 2021-05-27 by caternuson