adafruit/Adafruit-MCP23017-Arduino-Library #48: Interruptions - Method getLastInterruptPin() always returns 0 event if port 0 is not set for interruptions or event set up as input

izamoramx, 2020-07-08:
You might want to review the getLastInterruptPin() method. I tried a simple test using port B as input and port A as output modifying your interruptions sample code. Pin 0 (Port A) is set as output and I have an LED connected to it for debugging, and pin 8 (Port B) is set up as input with a button wired to it.

The interruptions correctly trigger when the state of pin 8 changes, however when polling the register for the interruption ports and values, it will always return pin 0 as the one that trigger the interruption. I don't see anything on the method that validates if the pin is set as an interruption pin or event as an input, but I don see the first thing the code checks if for the value of PortA but by bit, which explains why the method always stops at pin 0. I am not sure why the register if the MCP23017 is setting pin 0 as an interruption either since there are at least two places in the sample code that clear the INT register. If I comment out the block that checks for port A, the method correctly returns port 8 as the port that triggered the interruption.
 

Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

PsuFan, 2021-02-09:
I have seen this issue too. Root cause was a floating not connected input with a interrupt enabled from previous compile not being reset. This floating interrupt floods the chip/data line with messages making it impossible to even control outputs. Ensure all interrupted pins are high and add code to begin from PR #55 to disable previously interrupted pins or ensure chip is reset.

izamoramx, 2021-02-12:
@PsuFan thanks for the reply. I am not sure which code you are referring to. Maybe I missed it, if so please let me know. Otherwise it seems that the solution you are referring to has been implemented to the latest version of the library. If I understand this correctly, I will update my version to use this latest fix. Thanks again.

PsuFan, 2021-02-12:
Yes that is correct, see how the latest version works. A change to the interrupt example was also approved that ensures the interrupt is cleared.

Again ensure all interrupted pins are high (or low) as desired and not floating.

izamoramx, 2021-02-12:
Ok fantastic! 

You mean only the pins actually being used as INT correct? Not all the possible INT pins on the chip. I do have some pins that are part of the INT group but they are being used for other purposes. I would expect those wouldn't be affected.

PsuFan, 2021-02-12:
Yes the pins you define with mcp.setupInterruptPin(THIS_PIN,FALLING);

izamoramx, 2021-02-12:
Ok great. Thanks again! I will try that

caternuson, 2021-05-27:
@izamoramx Did the suggestions above resolve this?

izamoramx, 2021-06-23:
> @izamoramx Did the suggestions above resolve this?

@caternuson Sorry, I haven't had a chance to test the latest version of the library for this issue as I have to pause on that project to work on others but I am expecting to resume that in the next month or so and will be able to verify if the issue persists or not and get back to you.

caternuson, 2021-06-25:
OK, please also update to the 2.0.0 release of the library:
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/releases/tag/2.0.0
It changes things a lot.

caternuson, 2021-08-11:
Closing. Hoping fixed. Can reopen if still an issue.

closed 2021-08-11 by caternuson