adafruit/Adafruit-MCP23017-Arduino-Library #52: Inconsistent addressing

wmrichman, 2020-12-03:
I'm curious why i2cScanner shows two MCP23017 ICs (one with addresses jumpered all high, the other all low) as 0x20 and 0x27, but I have to use 0 and 7 in the "begin()" statements for it to work.  Using 0x20 and 0x27 compiled and sort of worked, but I ended up getting readings from the first chip for both addresses.  That seems inconsistent, as the other Adafruit libraries I've used seem to use the full i2c address.

Also, actually saying "begin(0)" results in an error; apparently since it's the default it's not allowed to be used?

PizzaProgram, 2020-12-15:
Did you try this working example first?  
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/issues/15#issuecomment-259647652

wmrichman, 2020-12-15:
Thanks.  I had gotten it working using single digit addresses.  I was just pointing out that it's inconsistent with the way Adafruit has done the addressing on their other parts.

ladyada, 2020-12-15:
yes this is an ancient library, one of the first arduino libraries in general :D so there were no standards then. we have a standard now, but dont want to break example code.

closed 2020-12-15 by ladyada