adafruit/Adafruit-MCP23017-Arduino-Library #58: Interrupt example is bad

lamello, 2021-03-27:
I have spend the day with the interrupt example and now i finally know whats the problem is.

In the example the commando is used:
mcp.setupInterruptPin(pin,FALLEN); 

This is really bad. The correct mode in this example is "CHANGE".
FALLING or RISING modes will not work with this code without instabilities.

What i see, is that with CHANGE a interrupt goes active with a change of the pin and the interrupt can be reset with readGPIOAB();
With FALLING it is different, The interrupt goes aktive when the pin goes LOW AND the interrupt STAYS active as long as the pin is low. 
While the pin is low the interrupt signal can't be reset !!!. 

This solved in the code by looping until the pin (or button) is released.

This has as consequent that the code will hang  ( no loop () ) until the user releases the button.

It would be nice that the examples have a standard that they always loop().

caternuson, 2021-06-16:
Yah, the MCP really only supports LOW/HIGH and not RISING/FALLING, since you can't clear the interrupt in that mode. It's a simple assert-if-comparison-true logic:

[image]

The changes in #64 include switching to LOW/HIGH.

caternuson, 2021-06-25:
Please checkout the new 2.0.0 release:
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/releases/tag/2.0.0
FALLING/RISING have been replaced with HIGH/LOW. CHANGE remains the same. These are (hopefully) inline with the Arduino spec:
https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/

lamello, 2021-06-27:
Thank you for the announcement.

The interrupt example is different, but is now good?
I don't think so. What's the value of an interrupt example which don't use the interrupt?
The interrupt example is to simplistic to be really helpful or practical.

caternuson, 2021-06-28:
What would you suggest? The example can be updated. Idea with current one is it's very simple - just shows how to generally configure the INTA/B pin to fire.

caternuson, 2021-07-13:
Closing due to lack of response. Can reopen if needed. Not really a library issue, just commentary about existing example.

closed 2021-07-13 by caternuson

kenttsai, 2022-08-05:
The example can be updated with an interrupt service routine so that it is more useful to general folks.