adafruit/Adafruit-MCP23017-Arduino-Library #61: Not an issue but suggestion for new library

richteel, 2021-05-12:
I used two MCP23017 ICs on a project with one for a 36 button keypad and a 32x122 graphic display. The chips worked fine but I wanted to improve the display updates by replacing one MCP23017 with a MCP23S17. I did not want to change my code too much so I rewrote the MCP23017 library for the MCP23S17. The library worked well as a drop in replacement. I would not mind if Adafruit picked up this library or create a new library to handle either IC. I searched the forum to see if I could find a way to submit suggestions for new libraries but did not come across how to submit library suggestions.

I would be willing to create a new library that allows the use of either MCP23017 or MCP23S17 but I would like to know if there is interest in Adafruit picking up the library.

My code is located at [https://github.com/richteel/TeelSys-MCP23S17-Arduino-Library](https://github.com/richteel/TeelSys-MCP23S17-Arduino-Library). If you take a look at the code, you can see there are only minor changes. It would be possible to combine into one library named MCP23X17.

Please let me know if there is any interest and the best way to help. I'm not looking for anything in return. I'm just hoping that I can help someone else out who starts out with one chip and would like to move to the other one.

richteel, 2021-05-12:
BTW: I did not include any example code as I was interested in getting my project working. The code that I'm currently working on using the library may be found at [https://github.com/richteel/Psion-Organiser-II---Graphic-LCD/tree/dev/Code/Keypad_LCD]https://github.com/richteel/Psion-Organiser-II---Graphic-LCD/tree/dev/Code/Keypad_LCD. The code is using the Adafruit library for the MCP23017 and my modified library for the MCP23S17.

caternuson, 2021-05-27:
@richteel This is essentially a request for MCP23S17 support? i.e. the SPI variant of the chip.

richteel, 2021-05-28:
Yes, I have written the code already but with a bit more work, both chips may be able to be folded into one library. Just a suggestion as it makes it easy to switch from one chip to the other.

caternuson, 2021-06-16:
@richteel Thanks for the offer. This will be hopefully be covered with #64 , which is a major re-write to attempt to address lots of currently outstanding issues and PRs.

caternuson, 2021-06-25:
OK, #64 has been merged and released:
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/releases/tag/2.0.0
Please give that a try. It includes support for both SPI and I2C variants of both 08 and 17 versions.

caternuson, 2021-06-28:
Closing for now. Can reopen if still an issue.

closed 2021-06-28 by caternuson

richteel, 2021-06-28:
@caternuson , Thanks, I will give this a try but it may be a few weeks before I can get to it.

caternuson, 2021-06-28:
@richteel And thanks for the offer to PR an update. It was just easier to roll that new support in along with all the other updates during the refresh.