adafruit/Adafruit-MCP23017-Arduino-Library #62: Error in order of statements in custom_frequency.ino example

karlsgodt, 2021-05-20:
- Arduino board:  Custom ATMEGA2560 board
- Arduino IDE version 1.8.12
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

In the Custom Frequency Example, **custom_frequency.ino** the two statements in the setup() function are out of order:

`Wire.setClock(1700000); // set frequency to 1.7mhz
`

...needs to come AFTER 

`mcp.begin(&Wire); // use default address 0
`

...or else the new clock frequency does not take effect.

ladyada, 2021-05-20:
ok! please submit a PR for the fix!

karlsgodt, 2021-05-20:
I need write access to the repo, yes?

caternuson, 2021-05-27:
@karlsgodt No, you'd make a pull request from your own copy (fork) on this repo. But I think #63 will also fix this eventually.

closed 2021-05-27 by caternuson