adafruit/Adafruit-MCP23017-Arduino-Library #65: MCP23018 support

Donkie, 2021-06-27:
Hi, good job on the rewrite, looks very good. Is there any plans on supporting the MCP23018 variant? It's pretty much the same as the MCP23017, with minor differences. I could use this library almost out of the box, the only issue I found so far is that this library does not support enabling the internal pull-up resistor for the OUTPUT mode, it only enables when choosing INPUT_PULLUP (which makes sense since MCP23017 only supports input pull-up.

The OUTPUT_OPEN_DRAIN mode should probably be accounted for as well then.

caternuson, 2021-06-28:
Maybe? The current base class is fairly specific to the push/pull variant registers. The differences seem small, but may require some more refactoring.

Donkie, 2021-06-28:
Alright. If you go ahead with this and need any help with testing on a real IC just hit me up.

caternuson, 2022-04-30:
FWIW, the CircuitPython library apparently "just works" with the caveat of needing to explicitly enable pull ups:
https://github.com/adafruit/Adafruit_CircuitPython_MCP230xx/issues/54