adafruit/Adafruit-MCP23017-Arduino-Library #69: MCP23017 v2.X Not Available?

czuvich, 2021-08-01:
I'm trying to grab the latest MCP23017 version (2.0.1) via the Arduino IDE (1.8.15 for Mac), and I only see up to 1.3.  Is 2.0.1 not available yet?  It looks like it should be.

caternuson, 2021-08-01:
I wonder if the name change here broke something with Library Manager?
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/pull/64/files#diff-f267f9ae2f5d5d56a1dbb305e95d40d4408d9ee7f71357f1f67fb18c29b7f082

ladyada, 2021-08-01:
@caternuson yah you cannot change the library name, plz change it back and re-release?

caternuson, 2021-08-02:
OK, looks like the rename fixed it. @czuvich Please check again.

[image]

czuvich, 2021-08-02:
Fixed!  Thanks for the quick update!

closed 2021-08-02 by czuvich