adafruit/Adafruit-MCP23017-Arduino-Library #71: Won't Compile

cstalsworth, 2021-08-07:
- Arduino board:  Uno/ProMini/ 

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8,13/1.8.15

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **None of the examples in 2.0.2 will compile* Previous versions compile correctly

Arduino: 1.8.15 (Windows 10), Board: "Arduino Uno"

Install 2.0.2 using library manager
select any example
compile

In file included from C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23X17.h:8:0,

                 from C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23X17.cpp:5:

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23X17.cpp: In member function 'uint16_t Adafruit_MCP23X17::readGPIOAB()':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23X17.cpp: In member function 'void Adafruit_MCP23X17::writeGPIOAB(uint16_t)':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp:23:0:

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp: In member function 'void Adafruit_MCP23XXX::pinMode(uint8_t, uint8_t)':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp: In member function 'uint8_t Adafruit_MCP23XXX::digitalRead(uint8_t)':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp: In member function 'void Adafruit_MCP23XXX::digitalWrite(uint8_t, uint8_t)':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp: In member function 'uint8_t Adafruit_MCP23XXX::readGPIO(uint8_t)':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp: In member function 'void Adafruit_MCP23XXX::writeGPIO(uint8_t, uint8_t)':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp: In member function 'void Adafruit_MCP23XXX::setupInterrupts(bool, bool, uint8_t)':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp: In member function 'void Adafruit_MCP23XXX::setupInterruptPin(uint8_t, uint8_t)':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp: In member function 'void Adafruit_MCP23XXX::disableInterruptPin(uint8_t)':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp: In member function 'uint8_t Adafruit_MCP23XXX::getLastInterruptPin()':

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: error: 'ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE' was not declared in this scope

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^

C:\MyPathDocuments\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.h:29:3: note: in definition of macro 'MCP23XXX_SPIREG'

   ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE //!< SPI register type

   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

CeruleanNutter, 2021-08-07:
I was having this issue earlier, updating to the latest version of the [Adafruit BusIO](https://github.com/adafruit/Adafruit_BusIO) library fixed it.
Currently MCP23017 version 2.0.2 is working with BusIO version 1.8.3 for me.

I wasn't even aware this library required BusIO; would be nice to have that stated somewhere on the README.md or in the example sketches.

caternuson, 2021-08-07:
The dependency on Adafruit_BusIO was added as part of #64. The requirement should be transparent for any new install of the MCP23017 library since it is called out in the `library.properties`
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/blob/63491b0aac7543868400c661124c0c9645d98cd7/library.properties#L10
That's why it's not mentioned in the README or examples. Ideally, it should just work.

I also just tested with a new install of 2.0.2 library and the `mcp23xxx_blink` example builds fine for a Arduino UNO target board. 

@cstalsworth Was this a fresh install of the library? It should have mentioned the library dependency and shown something like this:
[image]

cstalsworth, 2021-08-07:
No the dependency never popped up. I upgraded from 1.3 to 2.0.2. I had Bus IO 1.5 Installed so maybe that’s why. I upgraded BusIO and everything works. Thanks.

Chad Stalsworth

 

https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/blob/63491b0aac7543868400c661124c0c9645d98cd7/library.properties#L10

 <https://user-images.githubusercontent.com/8755041/128609774-e6939e15-ca87-4ed3-884f-56e6c10bc345.jpg>

caternuson, 2021-08-07:
It may be because of upgrade vs. fresh install also? Not sure if the Arduino Library Manager re-checks dependencies when updating. But glad it was just a matter of updating libs.

closed 2021-08-07 by caternuson