adafruit/Adafruit-MCP23017-Arduino-Library #73: mcp23S17 software addressing support

pawel3410, 2021-08-12:
For using couple of mcp23S17 is needet to connect with different CS pin for each expander. Is software addressing like in I2C version will be available? Than will be needed only one CS pin and set different addresses on pins A0,A1,A2.

pawel3410, 2021-08-12:
SPI version also support unique addressing. It's from datasheet:
3.3.2 ADDRESSING SPI DEVICES
(MCP23S17)
The MCP23S17 is a slave SPI device. The slave
address contains four fixed bits and three user-defined
hardware address bits (if enabled via IOCON.HAEN)

caternuson, 2021-08-12:
Oh yah, that. You can maybe give it a try? It's nominally supported by the BusIO register class:
https://github.com/adafruit/Adafruit_BusIO/blob/34f7392493718d311dc4ffccc745cc630ca7840f/Adafruit_BusIO_Register.h#L29-L34

pawel3410, 2021-08-13:
Unfortunately it is above my skills....

maziu, 2021-12-17:
@pawel3410 if you're still interested I implemented that in this PR https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/pull/78

caternuson, 2022-03-02:
Please try the 2.1.0 release of the library which includes @maziu 's PR #78 which adds this feature.
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/releases/tag/2.1.0

caternuson, 2022-03-25:
Closing due to lack of response. Should be fixed by PR mentioned above.

closed 2022-03-25 by caternuson