adafruit/Adafruit-MCP23017-Arduino-Library #76: Does not work with more than 2 MCP23017 on the same bus.

nicolasdeloigne, 2021-08-28:
Hi everyone,
I enjoy very much your library as it makes the use of a GPIO extender very friendly.
However, I encounter an issue. I manage to make the things work with up to 2 MCP23017 (with differents adresses), but as soon as I add one or more on the same bus, the things are not working (ie : the input are not going up when needed).
Did somebody already encounter a similar issue ?
Thanks in advance !

caternuson, 2021-08-28:
Run this I2C scanner as a sanity check:
https://playground.arduino.cc/Main/I2cScanner/
Do all addresses show up as expected?

nicolasdeloigne, 2021-08-28:
Yes, I run it, and all the adresses show up as expected (0x20 till 0x23).
The begin_I2C call on each declaration of the MCPs work also.
But after that, any call to digitalWrite(PIN_ID, HIGH) does not work.

WRoss7, 2021-08-28:
It worked before the last update, so I reverted to the previous version

nicolasdeloigne, 2021-08-28:
So, if I understand correctly, I have to remove the 2.0.2 version of the library to install the 2.0.1 downloading the zip in the Github archives ?

WRoss7, 2021-08-28:
Now that you mention this, when I downgraded I must have chosen the old version 1.2 from the library manager. however it works perfectly

[image]

caternuson, 2021-08-29:
Please provide more details on your setup.

Just tested 3x MCP23017's attached to a Qt PY with this example:
```cpp
#include <Adafruit_MCP23X17.h>

#define LED_PIN 7

Adafruit_MCP23X17 mcp1;
Adafruit_MCP23X17 mcp2;
Adafruit_MCP23X17 mcp3;

void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("MCP23017 Multiple Test!");

  mcp1.begin_I2C(0x20);
  mcp2.begin_I2C(0x21);
  mcp3.begin_I2C(0x22);

  mcp1.pinMode(LED_PIN, OUTPUT);
  mcp2.pinMode(LED_PIN, OUTPUT);
  mcp3.pinMode(LED_PIN, OUTPUT);

  Serial.println("Looping...");
}

void loop() {
  mcp1.digitalWrite(LED_PIN, HIGH);
  mcp2.digitalWrite(LED_PIN, HIGH);
  mcp3.digitalWrite(LED_PIN, HIGH);

  delay(500);
  
  mcp1.digitalWrite(LED_PIN, LOW);
  mcp2.digitalWrite(LED_PIN, LOW);
  mcp3.digitalWrite(LED_PIN, LOW);
  
  delay(500);
}
```
and it runs as expected.

caternuson, 2021-09-01:
Closing. Can't reproduce. Here's photo of test setup showing 3 MCP23017's blinking LEDs using sketch above.

[image]

closed 2021-09-01 by caternuson

nicolasdeloigne, 2021-09-02:
Thanks for your reply.
I am using 4 MCPs and looking to your sketch, I notice that you use one pair of pull-up resistors per MCP. What values do you use ?
I only use one pair on the complete bus lane, thinking that it is enough. Maybe the difference is here.

caternuson, 2021-09-02:
They were 4.7k. I still had this setup, so just tried with a single set of resistors as well. That also works.

caternuson, 2021-09-02:
This library was recently updated in a breaking way:
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/releases/tag/2.0.0
including adding a new dependency. So make sure all your libraries are up to date. If Adafruit BusIO is not installed, add that:
https://github.com/adafruit/Adafruit_BusIO

nicolasdeloigne, 2021-09-02:
Thank you for your answer ! 
The problem is solved, didn't came from the software but from a couple of cable that were not suitable for data transfer I suppose. I change them with a cable from an ethernet cable and it works smoothly !