adafruit/Adafruit-MCP23017-Arduino-Library #80: MCP23017 not working with Raspberry Pi Pico

sommerper, 2021-12-18:
First of all: Thanks for a great library that I have used with success on multiple occasions with Arduino Nano.
I'm trying to get this library to work on the Pico without success. I know the MCP23017 works because I have tested it with an Arduino Nano.

- Arduino board:  Raspberry Pi Pico

- Arduino IDE version (found in Arduino -> About Arduino menu):  Arduino IDE 1.8.18, Platformio 5.2.4, 3.4.0

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):

MCP23017 is connected via GP0 (SDA) and GP1 (SCL)

When I run the sketch below `mcp.begin_I2C()` always returns 0 and the sketch doesn't progress further. I suspect it has something to do with the Pico having 2 I2C busses?

```
#include <Arduino.h>
#include <Adafruit_MCP23X17.h>

Adafruit_MCP23X17 mcp;
int mcpLedPin = 0;

void setup()
{
  delay(2000);
  Serial.begin(9600);

  if (!mcp.begin_I2C())
  {
    Serial.println("Error.");
    while (1)
      ;
  }
  mcp.pinMode(mcpLedPin, OUTPUT);
}

void loop()
{
  mcp.digitalWrite(mcpLedPin, HIGH);
  delay(1000);
  mcp.digitalWrite(mcpLedPin, LOW);
  delay(1000);
}
```

caternuson, 2021-12-18:
Try GP4 for SDA and GP5 for SCL.

sommerper, 2021-12-19:
Hey!

I tried GP4 and GP5 which didn't work either. But you led me on to try the other I2C pins and it turns out GP6 (SDA1) and GP7(SLC1) work!

So thank you for leading on the right path.

Is there a way to define which pins the chip is connected to?
Something like
```
mcp.begin_I2C(9,10)
```
Then it would be possible to use the other I2C bus on the Pico.

Or maybe that is out of scope for this library?

Anyway, huge thanks to you!

caternuson, 2021-12-19:
The default pins are set by the core in the various variants files for the specific boards:
https://github.com/earlephilhower/arduino-pico/tree/master/variants
For other pins, can try pin reassignment:
https://arduino-pico.readthedocs.io/en/latest/

Closing since this is not a library issue.

closed 2021-12-19 by caternuson

sommerper, 2021-12-19:
Found the info, thanks. I appreciate you taking the time to help anyway.