adafruit/Adafruit-MCP23017-Arduino-Library #81: Version 2.0.2 does not compile with Teensy 4.1

sommerper, 2021-12-18:
- Arduino board:  Teensy 4.1

- Arduino IDE version:  Platformio 5.2.4, 3.4.0

- List the steps to reproduce the problem below:

Create a new project and add version 2.0.2.
Build the project and get the following output:

```
> Executing task in folder Teensy 4.1 MCP Error Project: platformio run --environment teensy41 <

Processing teensy41 (platform: teensy; board: teensy41; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.14.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy 1.155.0 (1.55) 
 - tool-teensy 1.155.0 (1.55) 
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 94 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit MCP23017 Arduino Library> 2.0.2
Building in release mode
Compiling .pio/build/teensy41/src/main.cpp.o
Compiling .pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23X08.cpp.o
Compiling .pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23X17.cpp.o
Compiling .pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23XXX.cpp.o
Compiling .pio/build/teensy41/FrameworkArduino/AudioStream.cpp.o
Compiling .pio/build/teensy41/FrameworkArduino/Blink.cc.o
Compiling .pio/build/teensy41/FrameworkArduino/CrashReport.cpp.o
Compiling .pio/build/teensy41/FrameworkArduino/DMAChannel.cpp.o
In file included from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.cpp:23:0:
.pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.h:8:37: fatal error: Adafruit_BusIO_Register.h: No such file or directory

*********************************************************************************
* Looking for Adafruit_BusIO_Register.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_BusIO_Register.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_BusIO_Register.h
*
*********************************************************************************

In file included from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23X17.h:8:0,
                 from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23X17.cpp:5:
.pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.h:8:37: fatal error: Adafruit_BusIO_Register.h: No such file or directory

*********************************************************************************
* Looking for Adafruit_BusIO_Register.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_BusIO_Register.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_BusIO_Register.h
*
*********************************************************************************

compilation terminated.
compilation terminated.
In file included from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23X08.h:8:0,
                 from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23X08.cpp:5:
.pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.h:8:37: fatal error: Adafruit_BusIO_Register.h: No such file or directory

*********************************************************************************
* Looking for Adafruit_BusIO_Register.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_BusIO_Register.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_BusIO_Register.h
*
*********************************************************************************

compilation terminated.
*** [.pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23X17.cpp.o] Error 1
*** [.pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23XXX.cpp.o] Error 1
*** [.pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23X08.cpp.o] Error 1
================================================================ [FAILED] Took 1.55 seconds ================================================================
The terminal process "platformio 'run', '--environment', 'teensy41'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Solution that works:
* Remove version 2.0.2 and install 1.2 instead.
* Then make sure to include Wire.h in the project

caternuson, 2021-12-18:
The Adafruit_BusIO library needs to be installed:
https://github.com/adafruit/Adafruit_BusIO

ladyada, 2021-12-18:
ping @ivankravetz

closed 2021-12-18 by ladyada

sommerper, 2021-12-19:
Thank you for the fast reply. Didn't expect that!

When I install Bus_IO v1.9.8 I get this:

```
> Executing task in folder Teensy 4.1 MCP Error Project: platformio run --environment teensy41 <

Processing teensy41 (platform: teensy; board: teensy41; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.14.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy 1.155.0 (1.55) 
 - tool-teensy 1.155.0 (1.55) 
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 94 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit MCP23017 Arduino Library> 2.0.2
Building in release mode
Compiling .pio/build/teensy41/src/main.cpp.o
Compiling .pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23X08.cpp.o
Compiling .pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23X17.cpp.o
Compiling .pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23XXX.cpp.o
Compiling .pio/build/teensy41/FrameworkArduino/AudioStream.cpp.o
Compiling .pio/build/teensy41/FrameworkArduino/Blink.cc.o
Compiling .pio/build/teensy41/FrameworkArduino/CrashReport.cpp.o
Compiling .pio/build/teensy41/FrameworkArduino/DMAChannel.cpp.o
In file included from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.cpp:23:0:
.pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.h:8:37: fatal error: Adafruit_BusIO_Register.h: No such file or directory

*********************************************************************************
* Looking for Adafruit_BusIO_Register.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_BusIO_Register.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_BusIO_Register.h
*
*********************************************************************************

In file included from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23X17.h:8:0,
                 from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23X17.cpp:5:
.pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.h:8:37: fatal error: Adafruit_BusIO_Register.h: No such file or directory

*********************************************************************************
* Looking for Adafruit_BusIO_Register.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_BusIO_Register.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_BusIO_Register.h
*
*********************************************************************************

compilation terminated.
compilation terminated.
In file included from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23X08.h:8:0,
                 from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23X08.cpp:5:
.pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.h:8:37: fatal error: Adafruit_BusIO_Register.h: No such file or directory

*********************************************************************************
* Looking for Adafruit_BusIO_Register.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_BusIO_Register.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_BusIO_Register.h
*
*********************************************************************************

[... 107 lines cut ...]

compilation terminated.
*** [.pio/build/teensy41/liba38/Adafruit BusIO/Adafruit_SPIDevice.cpp.o] Error 1
*** [.pio/build/teensy41/liba38/Adafruit BusIO/Adafruit_I2CDevice.cpp.o] Error 1
In file included from .pio/libdeps/teensy41/Adafruit BusIO/Adafruit_BusIO_Register.h:9:0,
                 from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.h:8,
                 from .pio/libdeps/teensy41/Adafruit MCP23017 Arduino Library/src/Adafruit_MCP23XXX.cpp:23:
.pio/libdeps/teensy41/Adafruit BusIO/Adafruit_I2CDevice.h:5:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

compilation terminated.
*** [.pio/build/teensy41/libb13/Adafruit MCP23017 Arduino Library/Adafruit_MCP23XXX.cpp.o] Error 1
================================================================ [FAILED] Took 2.92 seconds ================================================================
The terminal process "platformio 'run', '--environment', 'teensy41'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```

ladyada, 2021-12-19:
hmm this looks like a plaformio bug, please open an issue there as wire.h is a core library for i2c