adafruit/Adafruit-MCP23017-Arduino-Library #86: Provide access to INTCAP

caternuson, 2022-05-09:
There's already an existing (but stalled) PR that address this issue here:
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/pull/84
See that thread for more background information and discussion.

Opening this so there is a dedicated issue for tracking,

mastersta, 2022-07-31:
I'd like to throw in a vote to get this merged. Getting the INTCAP allows for using the MCP23XXX to read encoders.

ladyada, 2022-08-01:
@caternuson is that PR fine to merge?

caternuson, 2022-08-01:
@ladyada No. Not passing CI and no changes as yet based on feedback comments.

caternuson, 2022-08-02:
@mastersta Can you test the suggested code changes in PR #89?

mastersta, 2022-08-09:
@caternuson #89 tested and confirmed working as intended.

caternuson, 2022-08-09:
@mastersta excellent. thanks for testing!

caternuson, 2022-10-29:
Closing. Should be fixed with #89.

closed 2022-10-29 by caternuson