adafruit/Adafruit-MCP23017-Arduino-Library #90: Please include an example for using multiple mcp chips

businescat, 2022-10-16:
This is specific to the mcp23017 chip

#include <Adafruit_MCP23X17.h>

Adafruit_MCP23X17 mcp1; //declaration for chip 1
Adafruit_MCP23X17 mcp2; //declaration for chip 2

/*
 use the following table to configure A0-2 on your mcp chip to an I2c address
 
address  A2  A1  A0    i2c address
000     low low low     0x20         <-- we will be using this for chip 1
001     low low high    0x21         <-- and this one for chip 2
010     low high low    0x22             every additional address can be used for up to 8 chips total
011     low high high   0x23
100     high low low    0x24
101     high low high   0x25
110     high high low   0x26
111     high high high  0x27
*/
void setup() {
  Serial.begin(9600);
    mcp1.begin_I2C(0x20);   //call for a connection to chips using addresses above 
    mcp2.begin_I2C(0x21); 

  

  mcp1.pinMode(0, INPUT_PULLUP);  // table for the pin translation of mcp23 chips can be found on the main page of the library
  mcp1.pinMode(8, OUTPUT);
  mcp1.pinMode(9, OUTPUT);
  mcp2.pinMode(9, OUTPUT);
}

void loop() {
    
  Serial.print("pin0 = "); Serial.println(mcp1.digitalRead(0 ));

 if (mcp1.digitalRead(0) == 0){  // check if pin 0 is grounded, if so light up the led on pin 8 
  mcp1.digitalWrite(8, HIGH);
  }
  else{
  mcp1.digitalWrite(8, LOW);
 }
 
  mcp1.digitalWrite(9, HIGH);  // make chip 1 pin 9 led blink on and then off
  delay(500);
  mcp1.digitalWrite(9, LOW);
  delay(500);

  mcp2.digitalWrite(9, HIGH); // make chip 2 pin 9 led blink on and then off 
  delay(500);
  mcp2.digitalWrite(9, LOW);
  delay(500);

}

caternuson, 2022-10-16:
That looks like a proposed example sketch?

businescat, 2022-10-29:
Yes, I'm not a master coder or anything but it does work and shows how to read and write with multiple chips. I figured you would  want to make your own.

ryanmccarthypdx, 2023-11-10:
This is exactly what I'm looking for.  @businescat you should open a pull request into the repo with this.  If you are not familiar with that process, I'd be happy to do it for you.