adafruit/Adafruit-MCP23017-Arduino-Library #95: HELP - Multiple MCP23S17

MarcFinns, 2023-06-17:
- Arduino board:  Arduino Leonardo

- Arduino IDE version 1.8.9 

I built a custom board with two MCP23S17.
They are on the same Chip Select, with different addresses (000 and 111).
However, readGPIOAB returns the same for both chips, hence it appears I am not able to address them correctly. 
Reading the documentation i was under the impression this was a legit configuration.
Do they require different CS? If so, what is the value of having the address pins?
By the way, reset pins are connected to VCC. I did not implement a reset circuitry, which was working fine with a single IC.
Help appreciated!

bcollier94, 2023-12-27:
MarcFinns,

Don't know if you've figured it out or not, but after combing through another post the addresses are called by "0x20, 0x21, 0x22, etc."  Below is code I created for testing digital reads on one chip on the 0x21 address, and you can see I called its address with "mcp1.begin_I2C(0x21);"  For multiple, you would create more by calling "Adafruit_MCP23X17 mcpX;", where X is the different chips, so mcp1, mcp2, mcp3, etc.  In the previous version of the library I was able to call the addresses with just "0, 1, 2, etc," but that doesn't look to be the case anymore.

//------Begin code below------
#include <Wire.h>
#include <Adafruit_MCP23X17.h>

Adafruit_MCP23X17 mcp1;
uint8_t state[16];

void setup() {
  Serial.begin(115200);
  Wire.begin();

  mcp1.begin_I2C(0x21);

  for (int i = 0; i < 16; i++){
    mcp1.pinMode(i, INPUT_PULLUP);
    mcp1.digitalWrite(i, HIGH);
  }

}

void loop() {
  byte current[16];

  for (int b = 0; b < 16; b++){
    current[b] = mcp1.digitalRead(b);
    if (b == 15){
      Serial.println(current[b]);
    }else{
      Serial.print(current[b]);
    }
  }
  Serial.println();
}

MarcFinns, 2023-12-27:
> MarcFinns,
> Don't know if you've figured it out or not

Thanks, I did manage!
With it i built a MS Flight Simulator controller with encoders, buttons and switches. Works like a charm!