adafruit/Adafruit-MCP23017-Arduino-Library #96: GPIO7 bug hacking

siteswapjuggler, 2023-07-20:
Concerning this warning

Some people have reported an undocumented bug that can potentially corrupt the I2C bus. It occurs if an MCP230XX input pin state changes during I2C readout. This should be very rare. For more information, see this [forum post](https://www.microchip.com/forums/m646539.aspx) and this [knowledge base article](https://microchipsupport.force.com/s/article/On-MCP23008-MCP23017-SDA-line-change-when-GPIO7-input-change%3E).

It may appear that you can bypass this problematic by using the software interrupt and the interrupt capture register. Doing so you simulate a buffer that avoid the real time change bug which affect the SDA line. 

Just a thought by now that I've to test, but first I have to found a solution to replicate the bug efficiently.

caternuson, 2023-07-20:
Also posted here:
https://forums.adafruit.com/viewtopic.php?t=203060

siteswapjuggler, 2023-07-20:
The adafruit forum thread refer to the same problem but didn't mention any possible solution. That's I try to push with this issue.

siteswapjuggler, 2023-07-20:
I did also open a ticket with the microchip support in order to know if the other I2C members of the MCP230XX family like MCP23009 and MCP23018 are also affected or not.

Please notice that the MCP230XX start with all gpio as inputs so the bug can occur even if we don't use the 7th bit in our project. To avoid any problem you may have to set GPIOs 7 to output or activate internal pullup to avoid floating state on this gpio before any reading of the gpio state.

The other solution may be to never read the gpio state register but only use the interrupt capture register which is not affected by the real time change that cause the I2C crash.

siteswapjuggler, 2023-07-24:
I try to reproduce this I07 bug by feeding  a pwm signal into the GPIO7 of a MCP23008 and was unable to get any I2C crash. Btw I also tested the getCapturedInterrupt() method and it works great so definitively a good workout for this potential problem.

jokinpaha, 2024-08-09:
I think Microchip has circumvented this problem by making GPA7 and GPB7 output only for MCP23017 as it says in the datasheets first page and in pin descriptions.

siteswapjuggler, 2024-08-09:
They circumvented nothing they cover the mistake by announcing that GPx7 could not be used as input despite they are suppose to be built for this task.