adafruit/Adafruit-MCP23017-Arduino-Library #100: Not working using new Arduino Nano ESP32 development board

nestorcereno, 2024-03-05:
Not working using new Arduino Nano ESP32 development board (Arduino IDE).
Arduino Nano ESP32 development board is using ESP32-S3 MCU.

When compiling using ESP32-S3 board, there was no error but when using new Arduino Nano ESP32 board, there were errors.

Can somebody please update the library? Thank you in advance for your help.

caternuson, 2024-03-06:
>  using new Arduino Nano ESP32 board, there were errors.

What were the errors?

nestorcereno, 2024-03-06:
Thank you!

Here are the compilation results using ESP32 S3 Dev Module and Arduino nano ESP32 as board. I used the example code Adafruit MCP23017 Arduino Library (mcp23xxx)blink).

ESP32 S3 Dev Module as board: (compilation OK)
Sketch uses 287321 bytes (21%) of program storage space. Maximum is 1310720 bytes.
Global variables use 20304 bytes (6%) of dynamic memory, leaving 307376 bytes for local variables. Maximum is 327680 bytes.

Arduino nano ESP32 as board:
Thank you!

Here are the compilation results using ESP32 S3 Dev Module and Arduino nano ESP32 as board. I used the example code Adafruit MCP23017 Arduino Library (mcp23xxx)blink).

ESP32 S3 Dev Module as board: (compilation OK)
Sketch uses 287321 bytes (21%) of program storage space. Maximum is 1310720 bytes.
Global variables use 20304 bytes (6%) of dynamic memory, leaving 307376 bytes for local variables. Maximum is 327680 bytes.

Arduino nano ESP32 as board:
In file included from C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\cores\esp32/Arduino.h:223,
from C:\Users\chris\AppData\Local\Temp\arduino_build_162304\sketch\mcp23xxx_blink.ino.cpp:1:
C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\cores\esp32/io_pin_remap.h:47:61: error: 'digitalPinToGPIONumber' is not a type
#define pinMode(pin, mode) pinMode(digitalPinToGPIONumber(pin), mode)
^~~~~~~~~~~~~~~~~~~~~~
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src/Adafruit_MCP23XXX.h:50:8: note: in expansion of macro 'pinMode'
void pinMode(uint8_t pin, uint8_t mode);
^~~~~~~
C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\cores\esp32/io_pin_remap.h:42:65: error: 'digitalPinToGPIONumber' is not a type
#define digitalRead(pin) digitalRead(digitalPinToGPIONumber(pin))
^~~~~~~~~~~~~~~~~~~~~~
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src/Adafruit_MCP23XXX.h:51:11: note: in expansion of macro 'digitalRead'
uint8_t digitalRead(uint8_t pin);
^~~~~~~~~~~
C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\cores\esp32/io_pin_remap.h:46:66: error: 'digitalPinToGPIONumber' is not a type
#define digitalWrite(pin, val) digitalWrite(digitalPinToGPIONumber(pin), val)
^~~~~~~~~~~~~~~~~~~~~~
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src/Adafruit_MCP23XXX.h:52:8: note: in expansion of macro 'digitalWrite'
void digitalWrite(uint8_t pin, uint8_t value);
^~~~~~~~~~~~
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\examples\mcp23xxx_blink\mcp23xxx_blink.ino: In function 'void setup()':
C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\cores\esp32/io_pin_remap.h:47:83: error: invalid conversion from 'int8_t' {aka 'signed char'} to 'int ()(uint8_t)' {aka 'int ()(unsigned char)'} [-fpermissive]
#define pinMode(pin, mode) pinMode(digitalPinToGPIONumber(pin), mode)
^
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\examples\mcp23xxx_blink\mcp23xxx_blink.ino:30:7: note: in expansion of macro 'pinMode'
mcp.pinMode(LED_PIN, OUTPUT);
^
C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\cores\esp32/io_pin_remap.h:47:61: note: initializing argument 1 of 'void Adafruit_MCP23XXX::pinMode(int (*)(uint8_t), uint8_t)'
#define pinMode(pin, mode) pinMode(digitalPinToGPIONumber(pin), mode)
^~~~~~~~~~~~~~~
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src/Adafruit_MCP23XXX.h:50:8: note: in expansion of macro 'pinMode'
void pinMode(uint8_t pin, uint8_t mode);
^~~~~~~
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\examples\mcp23xxx_blink\mcp23xxx_blink.ino: In function 'void loop()':
C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\cores\esp32/io_pin_remap.h:46:88: error: invalid conversion from 'int8_t' {aka 'signed char'} to 'int ()(uint8_t)' {aka 'int ()(unsigned char)'} [-fpermissive]
#define digitalWrite(pin, val) digitalWrite(digitalPinToGPIONumber(pin), val)
^
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\examples\mcp23xxx_blink\mcp23xxx_blink.ino:36:7: note: in expansion of macro 'digitalWrite'
mcp.digitalWrite(LED_PIN, HIGH);
^~
C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\cores\esp32/io_pin_remap.h:46:66: note: initializing argument 1 of 'void Adafruit_MCP23XXX::digitalWrite(int (*)(uint8_t), uint8_t)'
#define digitalWrite(pin, val) digitalWrite(digitalPinToGPIONumber(pin), val)
^~~~~~~~~~~~~~~~~~~
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src/Adafruit_MCP23XXX.h:52:8: note: in expansion of macro 'digitalWrite'
void digitalWrite(uint8_t pin, uint8_t value);
^~~~~~~~~~~~
C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\cores\esp32/io_pin_remap.h:46:88: error: invalid conversion from 'int8_t' {aka 'signed char'} to 'int ()(uint8_t)' {aka 'int ()(unsigned char)'} [-fpermissive]
#define digitalWrite(pin, val) digitalWrite(digitalPinToGPIONumber(pin), val)
^
C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\examples\mcp23xxx_blink\mcp23xxx_blink.ino:38:7: note: in expansion of macro 'digitalWrite'
mcp.digitalWrite(LED_PIN, LOW);
^~
[... cut ...]
Using library Adafruit_MCP23017_Arduino_Library at version 2.3.2 in folder: C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library
Using library Adafruit_BusIO at version 1.14.1 in folder: C:\Users\chris\OneDrive\Documents\Arduino\libraries\Adafruit_BusIO
Using library Wire at version 2.0.0 in folder: C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\libraries\Wire
Using library SPI at version 2.0.0 in folder: C:\Users\chris\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\libraries\SPI
exit status 1
Error compiling for board Arduino Nano ESP32.

nestorcereno, 2024-03-06:
I tried the IDE Tools option "Pin numbering: By GPIO number (legacy)" and the compilation was OK!

I will try if the example code will work when i upload it to the new Arduino Nano ESP32 code.

Thank you!

caternuson, 2024-03-07:
Thanks for the compilation output. Was able to recreate this locally.

The issue is with the ESP32 Board Support Package. There's an open issue about this here:
https://github.com/espressif/arduino-esp32/issues/9150

I tried the suggested "easy out" option:

> By forcing users to use the standard ESP32 pin naming (by choosing "Tools" -> "Pin Numbering" -> "By GPIO number (legacy)" in the Arduino IDE), no macros will be defined and your library will work out-of-the-box with no source changes.

and that got rid of the compilation errors, however, did not test any further (just a compilation check).

Try that and see if it both compiles and runs as expected once the sketch is uploaded.

LuGr-01, 2024-10-20:
Hey there, I do not have such an option in platform.io. Do you know any other solutions for this?

[image]